adafruit/Adafruit_FXAS21002C #4: Range Selection not implemented

Wind4Greg, 2017-10-31:
The `begin(gyroRange_t rng)` function of the driver takes a gyroscope range parameter (+/- 2000, 1000, 500, or 250 degrees per second). This doesn't actually doesn't get sent to the FXAS21002C.

To properly implement this `CTRL_REG0` needs to be set (or so it seems to me). 

Here is the fix that I am using:

```
/**************************************************************************/
/*!
    @brief  Setups the HW
*/
/**************************************************************************/
bool Adafruit_FXAS21002C::begin(gyroRange_t rng)
{
  /* Enable I2C */
  Wire.begin();

  /* Set the range the an appropriate value */
  _range = rng;

  /* Clear the raw sensor data */
  raw.x = 0;
  raw.y = 0;
  raw.z = 0;

  /* Make sure we have the correct chip ID since this checks
     for correct address and that the IC is properly connected */
  uint8_t id = read8(GYRO_REGISTER_WHO_AM_I);
  // Serial.print("WHO AM I? 0x"); Serial.println(id, HEX);
  if (id != FXAS21002C_ID)
  {
    return false;
  }

  /* Set CTRL_REG1 (0x13)
   ====================================================================
   BIT  Symbol    Description                                   Default
   ---  ------    --------------------------------------------- -------
     6  RESET     Reset device on 1                                   0
     5  ST        Self test enabled on 1                              0
   4:2  DR        Output data rate                                  000
                  000 = 800 Hz
                  001 = 400 Hz
                  010 = 200 Hz
                  011 = 100 Hz
                  100 = 50 Hz
                  101 = 25 Hz
                  110 = 12.5 Hz
                  111 = 12.5 Hz
     1  ACTIVE    Standby(0)/Active(1)                                0
     0  READY     Standby(0)/Ready(1)                                 0
  */
  /* Set CTRL_REG0 (0x0D)  Default value 0x00
  =====================================================================
  BIT  Symbol     Description                                   Default
  7:6  BW         cut-off frequency of low-pass filter               00 
    5  SPIW       SPI interface mode selection                        0
  4:3  SEL        High-pass filter cutoff frequency selection        00
  2    HPF_EN     High-pass filter enable                             0
  1:0  FS         Full-scale range selection
                  00 = +-2000 dps
                  01 = +-1000 dps
                  10 = +-500 dps
                  11 = +-250 dps
  The bit fields in CTRL_REG0 should be changed only in Standby or Ready modes.
  */
  uint8_t ctrlReg0 = 0x00;
  switch(_range)
  {
    case GYRO_RANGE_250DPS:
      ctrlReg0 = 0x03;
      break;
    case GYRO_RANGE_500DPS:
     ctrlReg0 = 0x02;
      break;
    case GYRO_RANGE_1000DPS:
      ctrlReg0 = 0x01;
      break;
    case GYRO_RANGE_2000DPS:
      ctrlReg0 = 0x00;
      break;
  }

  /* Reset then switch to active mode with 100Hz output */
  write8(GYRO_REGISTER_CTRL_REG1, 0x00); // Standby
  write8(GYRO_REGISTER_CTRL_REG1, (1<<6)); // Reset
  write8(GYRO_REGISTER_CTRL_REG0, ctrlReg0); // Set sensitivity
  write8(GYRO_REGISTER_CTRL_REG1, 0x0E); // Active
  delay(100); // 60 ms + 1/ODR
  /* ------------------------------------------------------------------ */

  return true;
}
```
Here is a gzip of the modified file. The `*.h` file did not need modification.
[Adafruit_FXAS21002C.cpp.gz](https://github.com/adafruit/Adafruit_FXAS21002C/files/1432602/Adafruit_FXAS21002C.cpp.gz)

Cheers
Greg B.

ekalyvio, 2017-11-21:
I agree with that! I literally spent 6 hours of why my custom made source code (under esp-open-rtos) was giving me much worst results compared to the Adafruit's one. They have to fix that...

microbuilder, 2017-11-21:
Thanks for the heads up ... I just pushed version 1.1.3 out with the following fix: https://github.com/adafruit/Adafruit_FXAS21002C/commit/b63cd3b29608f8d217275c4accae8e791e832983

closed 2017-11-21 by microbuilder