adafruit/Adafruit_FXAS21002C #7: Scaled output for different sensitivity ranges

Farsad, 2019-08-02:
This library return scaled values for different ranges. As an example, in a similar movement gyro z value is ~0.50rad/s for 250DPS but ~6rad/s for 2000DPS.

I've tested it with different programs and examples include https://github.com/adafruit/Adafruit_FXAS21002C/blob/master/examples/sensorapi/sensorapi.ino . 

These are the values I got from the sensorapi.ino app for an almost similar rotation around z.

**GYRO_RANGE_250DPS**

X: 0.03 Y: -0.00 Z: 0.01 rad/s 
X: -0.07 Y: 0.02 Z: -0.42 rad/s 
X: 0.14 Y: -0.11 Z: **0.90** rad/s 
X: -0.01 Y: 0.01 Z: -0.03 rad/s 
X: -0.02 Y: 0.05 Z: -0.42 rad/s 
X: 0.26 Y: 0.04 Z: **0.98** rad/s 
X: -0.03 Y: 0.04 Z: -0.08 rad/s 
X: -0.06 Y: 0.06 Z: -0.13 rad/s 
X: 0.17 Y: 0.00 Z: **0.93** rad/s 
X: -0.03 Y: 0.03 Z: -0.13 rad/s 
X: -0.01 Y: 0.06 Z: -0.10 rad/s 
X: 0.28 Y: 0.04 Z: **1.01** rad/s 
X: -0.04 Y: 0.05 Z: -0.12 rad/s 
X: 0.05 Y: 0.01 Z: -0.22 rad/s 
X: 0.19 Y: 0.00 Z: **1.06** rad/s 

**GYRO_RANGE_500DPS**

X: 0.02 Y: -0.25 Z: -0.41 rad/s 
X: -0.10 Y: 0.13 Z: **1.76** rad/s 
X: -0.05 Y: 0.00 Z: -0.22 rad/s 
X: 0.01 Y: 0.04 Z: -0.02 rad/s 
X: 0.65 Y: 0.68 Z: **2.68** rad/s 
X: -0.10 Y: 0.08 Z: -0.44 rad/s 
X: -0.02 Y: -0.01 Z: 0.04 rad/s 
X: -0.09 Y: 0.05 Z: 0.60 rad/s 
X: -0.24 Y: 0.06 Z: -2.29 rad/s 
X: 0.23 Y: -0.11 Z: **1.82** rad/s 
X: 0.00 Y: -0.10 Z: -0.21 rad/s 
X: 0.12 Y: 0.01 Z: -0.12 rad/s 
X: -0.33 Y: 0.38 Z: **1.88** rad/s 
X: -0.01 Y: -0.03 Z: -1.09 rad/s 
X: 0.05 Y: -0.03 Z: 0.07 rad/s 

**GYRO_RANGE_2000DPS**

X: 0.33 Y: 1.29 Z: **4.90** rad/s 
X: -0.65 Y: -0.36 Z: -1.91 rad/s 
X: 0.23 Y: -0.13 Z: 0.09 rad/s 
X: -2.06 Y: 0.74 Z: **6.31** rad/s 
X: 0.40 Y: -2.00 Z: -11.05 rad/s 
X: 0.14 Y: -0.39 Z: **3.32** rad/s 
X: -0.23 Y: -0.08 Z: -0.33 rad/s 
X: 1.06 Y: 0.25 Z: 0.34 rad/s 
X: 2.03 Y: 3.46 Z: **9.14** rad/s 
X: -0.38 Y: -0.35 Z: -6.22 rad/s 
X: 0.11 Y: -0.12 Z: 0.51 rad/s 
X: 0.41 Y: -0.29 Z: -0.29 rad/s 
X: -0.50 Y: -1.13 Z: -2.07 rad/s 
X: 1.39 Y: 2.13 Z: **7.12** rad/s 
X: -0.21 Y: -0.18 Z: -0.57 rad/s 
X: 0.23 Y: -0.25 Z: 0.39 rad/s 
X: -2.27 Y: 1.29 Z: **7.35** rad/s 
X: 1.32 Y: -1.82 Z: -13.31 rad/s 
X: 0.02 Y: -0.29 Z: 2.85 rad/s 
X: 0.10 Y: -0.24 Z: -0.56 rad/s 
X: 0.30 Y: -0.23 Z: 0.24 rad/s 
X: -0.14 Y: 3.33 Z: **10.46** rad/s 
X: -1.44 Y: -0.91 Z: -7.65 rad/s

We have tested it with three sensors on installed on three different Adafruit Feather 32u4 Adalogger and the behavior is the same for all of them.

There is a thread on the forum for your reference: https://forums.adafruit.com/viewtopic.php?f=19&t=154771&p=763515#p763515

ladyada, 2019-08-02:
ooh - ok - want to try and find the bug location and submit a PR? that will get it fixed fastest!

Farsad, 2019-08-03:
Hi, I'm not an expert in C but just took a look at the codes and couldn't find any problem. My only guess is there is an issue in setting the range on the device but the runtime code doesn't recognize it and multiplies the values base on the selected range.

ladyada, 2019-08-03:
@siddacious take a look!

Farsad, 2019-08-03:
@siddacious Hi, Do you have any quick suggestion to fix the recorded data for now?

siddacious, 2019-08-03:
Hi @Farsad, I agree with your assessment that the range setting isn't working as expected but the library isn't aware and scales the data incorrectly. The _quicker_ though not quick fix for the data would be to figure out which data rate is _actually_ being used and only specify that one so the scaling is applied properly.

I'm going to throw out a guess that it's using the default of 2000DPS, but short of comparing the measured values to a reference or estimating which reading is closest to reality, the only way to know for sure is to debug the driver to figure out what is being set (or not).

Farsad, 2019-08-03:
@siddacious, Thanks for your reply I just did some tests and. Updating my gyro data by Gyroscope_value/0.0078125*0.0625  gives a good result. so, I guess you are right about the default 2000DPS setting.

Farsad, 2019-08-04:
@siddacious, I found that I had an old version of this Library. I'm not sure how I got that version, I was check and see the latest code changes here is for 2 years ago. So I thought that I should have the latest version of it. Sorry for that, I confirm the issue is fixed before and does not exist on the latest version of this library.  :)

closed 2019-08-04 by Farsad