adafruit/Adafruit_FXAS21002C #13: Range set was seems to have been removed for some reason

calisto27, 2021-08-05:
The ability to set the gyro range in .begin() that was implemented in version 1.13 (see issue #4) no longer exists in the latest version.

caternuson, 2021-08-05:
Yep, that was changed as part of #11, which was known to be a breaking change.

This would probably be best supported via dedicated get/set functions. Something like `setGyroRange()`. Would that work? Or is there some driving reason to have the gyro range be a parameter to `begin()`?

calisto27, 2021-08-05:
I just get the error message 'class Adafruit_FXAS21002C' has no member named 'setGyroRange'

calisto27, 2021-08-05:
I suppose it doesnt matter how it is set.
Currently it seems that the only way to change the gyro is in the .ccp file itself.

caternuson, 2021-08-05:
Yep, the function isn't implemented yet.

calisto27, 2021-08-05:
Great, thanks

caternuson, 2021-08-05:
Reopening. We can leave this open until the pull request gets merged and the changes released. That way you can test the changes and make sure they work for your setup. There may still be something needed? Hopefully not, seems like a simple fix. But OK to leave this open until we know for sure.

closed 2021-08-07 by calisto27